NVIDIA/warp #3: nvrtc: error: failed to open nvrtc-builtins64_113.dll

jlinBE, 2022-03-23:
I got this error message when running "python -m warp.tests":

nvrtc: error: failed to open nvrtc-builtins64_113.dll

environment: Win 10 x64; CUDA toolkit 11.5; Python 3.8.7; Nvidia graphics driver 496.13; Visual Studio 2019

steps to reproduce: 1. download [v0.1.25-alpha](https://github.com/NVIDIA/warp/releases/tag/v0.1.25-alpha); 2. "pip install ." ; 3. "python -m warp.tests". 4. error.

The .dll file in question is found in both the warp project directory, and also in the python install directory:

"C:\Users\<my username>\AppData\Local\Programs\Python\Python38\Lib\site-packages\warp\bin\nvrtc-builtins64_113.dll"

So maybe it's just a pathing issue - any pointers?

Screenshot attached.
[image]

mmacklin, 2022-03-23:
Thanks for the detailed report @jlinBE- we've seen this happen on a couple of users machines but haven't been able to narrow it down. 

The `/bin` path should be added to the system path through this code:

https://github.com/NVIDIA/warp/blob/main/warp/context.py#L452

There is a special path for Python 3.8.x that uses `os.add_dll_directory()`, which I suspect may be implicated here. Will do some more investigation into this, but if you find any workaround please let us know.

mmacklin, 2022-03-24:
Hi @jlinBE , I have pushed an update that may address your issue (using non-OS path characters):

https://github.com/NVIDIA/warp/commit/ae8717fc9988354527cff19800a5322f827da83a

I have updated the release package with this change. Please let us know if it resolves things for you.

jlinBE, 2022-03-25:
hi @mmacklin, the same mistake (using + to concatenate paths instead of os.path.join()) is repeated frequently in context.py, and you only fixed a single instance.  In fact you should probably check the rest of the project too.

You were on the right track, though - I changed a few of those on my end, and the tests ran.

mmacklin, 2022-03-25:
OK, thanks seems like this is the culprit. We'll go through and address the other instances.

mmacklin, 2022-03-28:
Hi @jlinBE, I have updated the use of `os.path.join()` throughout the codebase in this commit: https://github.com/NVIDIA/warp/commit/bdb8ac7f1491ecfa5386ae4779415a68b7e2776a,

I have updated the release package with this change, please let me know if addresses the issue for you.

jlinBE, 2022-03-28:
yes - it works now.  thanks!

closed 2022-03-28 by jlinBE