NVIDIA/warp #5: parse_urdf errors

cadop, 2022-03-23:
In `parse_urdf()` there is a call to `urdfpy.urdf.load`, I think this is supposed to be `urdfpy.URDF.load`. 

There is also `builder.add_link` but this gives the error `AttributeError: 'ModelBuilder' object has no attribute 'add_link'`, which I guess is true since I don't see it.

mmacklin, 2022-03-24:
Thanks for the report, the sim. package is under quite heavy development but will look to address this in a future version. Mostly we have focused on MJCF import.

psh117, 2022-05-15:
I have a different problem in `wp.sim.parse_urdf`  I got another problem while loading URDF collision mesh. `wp.sim.urdf_add_collision()`

The traceback is  
```python
Traceback (most recent call last):
  File "urdf_test.py", line 194, in <module>
    robot = Robot(render=True, device=wp.get_preferred_device(), num_envs=1)
  File "urdf_test.py", line 49, in __init__
    wp.sim.parse_urdf(os.path.join(os.path.dirname(__file__), "panda_arm.urdf"), builder,
  File "/home/dyros/anaconda3/envs/suhan38/lib/python3.8/site-packages/warp/sim/import_urdf.py", line 166, in parse_urdf
    urdf_add_collision(
  File "/home/dyros/anaconda3/envs/suhan38/lib/python3.8/site-packages/warp/sim/import_urdf.py", line 91, in urdf_add_collision
    mesh = wp.Mesh(vertices, faces)
  File "/home/dyros/anaconda3/envs/suhan38/lib/python3.8/site-packages/warp/types.py", line 727, in __init__
    if (points.device != indices.device):
AttributeError: 'list' object has no attribute 'device'
```

I think this is because the `faces` and `vertices` are just python array not `wp` objects.  

```python
        if geo.mesh:

            for m in geo.mesh.meshes:
                faces = [] # <-- array
                vertices = [] # <-- array

                for v in m.vertices:
                    vertices.append(np.array(v))

                for f in m.faces:
                    faces.append(int(f[0]))
                    faces.append(int(f[1]))
                    faces.append(int(f[2]))

                mesh = wp.Mesh(vertices, faces) # <-- error occurs here 

                builder.add_shape_mesh(
                    link,
                    pos,
                    rot,
                    mesh,
                    density,
                    ke=shape_ke,
                    kd=shape_kd,
                    kf=shape_kf,
                    mu=shape_mu)
```

cadop, 2022-05-17:
@psh117 

The solution is to update the urdf_loader on line 95 to:

`mesh = wp.sim.model.Mesh(vertices, faces)`

Because of warp using `import *`, its import resolution is picking up the `type.Mesh` as `model.Mesh`, and thats what I was showing above.  So you need to explicitly refer to the sim package Mesh.

mmacklin, 2022-06-16:
Fixed in https://github.com/NVIDIA/warp/commit/419e592017135480f5d55791d726b726c4ffa4ca.

closed 2022-06-16 by mmacklin