NVIDIA/warp #8: Failed to open USD file

FantasyVR, 2022-03-24:
When running examples, the errors show that I can't open USD files. I've installed `usd-core`.
```
(warp) xx@xx:~/code/warp$ python examples/example_mesh.py 
Warp initialized:
   Version: 0.1.25
   Using CUDA device: GeForce RTX 2060
   Using CPU compiler: /usr/lib/ccache/g++
Traceback (most recent call last):
  File "example_mesh.py", line 103, in <module>
    usd_stage = Usd.Stage.Open(os.path.join(os.path.dirname(__file__), "assets/bunny.usd"))
pxr.Tf.ErrorException: 
        Error in 'pxrInternal_v0_22__pxrReserved__::UsdStage::Open' at line 878 in file /opt/USD/pxr/usd/usd/stage.cpp : 'Failed to open layer @assets/bunny.usd@'
```

mmacklin, 2022-03-24:
Hmm, I haven't seen this before. Could you check that the file `examples/assets/bunny.usd` exists on your HD?

Could you also verify that the path string: `os.path.join(os.path.dirname(__file__), "assets/bunny.usd")` is valid, and finally one more thing to try would be running directly from inside the `examples` working directory.

etaoxing, 2022-03-25:
Also ran into this error because the files didn't exist. Those files are stored with git-lfs, so needed to `apt-get install git-lfs` and then reclone the repo. Would be nice to add a note in the README about this.

mmacklin, 2022-03-25:
Thanks @etaoxing, I updated the README to recommend LFS installation before cloning.

closed 2022-03-25 by mmacklin