NVIDIA/warp #13: can't import after pip install

pieper, 2022-03-25:
I tried pip install warp as described [in this blog post](https://developer.nvidia.com/blog/creating-differentiable-graphics-and-physics-simulation-in-python-with-nvidia-warp/?mkt_tok=MTU2LU9GTi03NDIAAAGDXGPha9e1ZJkTVeEFCVQv4ZUaERtdbwwVzcUXotTCsiQyDMLqTgpfgQG0C9bRvJnstGfg80C9u5YzTHPVoTHNF2mFXiTF8jC_E_wMjVUsXPnrISG2) but even after it says it was installed I get this:
```
ubuntu@ip-172-31-31-7:/mnt/extra/pieper/SlicerDMRI$ pip3 install warp
Collecting warp
  Using cached warp-0.0.1-py3-none-any.whl (1.1 kB)
Installing collected packages: warp
Successfully installed warp-0.0.1
ubuntu@ip-172-31-31-7:/mnt/extra/pieper/SlicerDMRI$ python3
Python 3.8.10 (default, Nov 26 2021, 20:14:08) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import warp
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'warp'
```

This is ubuntu 20.04 with the standard python and pip packages.

mmacklin, 2022-03-25:
Hi @pieper , I think the instructions in the blog are slightly unclear. You need to first download the Warp package from the Github releases section on this repository, then follow the README.md instructions.

What happened here is that `pip install warp` will install an unrelated package from PyPi with the same name.

I will update the blog to point to the full install instructions.

Thanks,
Miles

pieper, 2022-03-25:
Thanks @mmacklin - if I install from the github release it installs and the tests run.

closed 2022-03-25 by pieper