NVIDIA/warp #14: adjoint=False ignored in launch() for some data types?

varkenvarken, 2022-03-25:
In a simple kernel that multiplies an array of matrices with an array of vectors I get a CUDA compilation error when using an array of mat44 and arrays of vec4

    error: no suitable constructor exists to convert from "void" to "wp::mat44"

This does not happen if I define and launch a similar kernel with mat33 and vec3 

I have attached both the working 3x3 and the non-working 4x4 Python  samples along with the full error text

[files.zip](https://github.com/NVIDIA/warp/files/8352150/files.zip)

To replicate simply run

    python test_works.py

and

    python test.py

The last one gives the error. 

extra info:
```
Warp Version: 0.1.25   Using CUDA device: NVIDIA GeForce GTX 1080
NVIDIA-SMI 495.29.05    Driver Version: 495.29.05    CUDA Version: 11.5
```

varkenvarken, 2022-03-25:
and I forgot: this was on Ubuntu 20.04.4 LTS

mmacklin, 2022-03-28:
Hi @varkenvarken, thanks for the report - I fixed a bug with mat44 adjoint in this commit:

https://github.com/NVIDIA/warp/commit/b7d99d1ee2b4cd5f7984addd312cb566c27b4999

I've updated the release packages to include this, let me know if it works for you.

Just a couple of comments on your benchmark code, there is a `wp.ScopedTimer()` class you may want to use to make timing easier. This profiles CPU time currently, so you should still have a `wp.synchronize()` inside the scope in case you want a measure GPU time.

Copying the data back to NumPy will likely dominate the time in this benchmark. If you just want to know the kernel time I would recommend moving the call to `array.numpy()` outside the timer scope. For larger applications it can also make sense to perform the copy back to host asynchronously, using the `wp.copy()` method back to a CPU array.

varkenvarken, 2022-03-28:
Thanks Miles, works perfectly. I saw that the commit touched a couple of other data types as well and I haven't tested those yet but if I encounter something I will raise an issue.

-- Michel.

PS> and thanks for the ScopedTimer() tip!

closed 2022-03-29 by mmacklin