NVIDIA/warp #15: make the non-commercial clause more visible in the readme

pieper, 2022-03-26:
This looks like an interesting library, but after reviewing the license I realized it says it's for non-commercial use only.  I think that information should be made more clear so people don't invest time unless they are comfortable with that limitation.  (Of course an even better solution would be to change the license to something like Apache 2, in case you are open to that option.)

mmacklin, 2022-03-28:
Hi @pieper, thanks for feedback - I've added a section in the README.md to point people to the license. We are evaluating more permissive licenses for future versions.

Cheers,
Miles

pieper, 2022-03-28:
Thanks for the edit @mmacklin.  Just for clarity I suggest some extra wording if you don't mind.

https://github.com/NVIDIA/warp/pull/17

mmacklin, 2022-05-22:
Hi @pieper, apologies for the delay - I added the suggested note to the README.md.

Thanks,
Miles

closed 2022-05-22 by mmacklin