NVIDIA/warp #24: FileNotFoundError: Could not find module 'warp.dll'

DesmondZhong, 2022-05-04:
After building from source, I encounter the following error when running tests. 

```
FileNotFoundError: Could not find module 'warp.dll' (or one of its dependencies). Try using the full path with constructor syntax.
```

After spending some time on the problem, I figured out that this is due to a bug of Python 3.8 on Windows. 

This stackoverflow post pointed out the issue.
https://stackoverflow.com/questions/59330863/cant-import-dll-module-in-python/64472088#64472088

The author of the above post has reported this bug and it has been fixed for future python versions. 
https://github.com/python/cpython/issues/86280

In short, the solution is to explicitly pass `winmode=0` to `ctypes.CDLL`. In Python 3.8, the default value of `winmode` is set to `None`, which is unintended as it is not consistent with the documentation. This argument only affects the loading logic under Windows. 

Besides this change, I also need to pass the full path of the file `warp.dll` instead of just the filename in order to successfully load the file. This is suggested in the error message above.

I'll be creating a pull request to fix this issue soon. I think python 3.8 users on Windows will encounter this problem and will benefit from the fix. 

------
**My environment**
- OS: Windows 10
- Python version: 3.8.10
- MSVC Compiler Version (cl.exe version) 19.26.28806 for x64
- Warp version: 0.1.25

mmacklin, 2022-06-16:
Hi Desmond, I think we finally fixed this properly using your suggestion to modify the winmode in this change, https://github.com/NVIDIA/warp/commit/0fbe6aee7a742cac56b0087bc004eda8284e7ad1.

Thanks,
Miles

closed 2022-06-16 by mmacklin

cadop, 2022-06-18:
I'm not sure this is the same problem, but Windows 10 Python 3.9 I now have this error. 

`pip install numpy`
`pip install usd-core`
`pip install warp-lang`
`python -m warp.tests`

```
Traceback (most recent call last):
  File "C:\Program Files\Python39\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Program Files\Python39\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\...\venv_warp\lib\site-packages\warp\tests\__main__.py", line 1, in <module>
    from . test_all import run
  File "C:\...\venv_warp\lib\site-packages\warp\tests\test_all.py", line 13, in <module>
    import warp.tests.test_codegen
  File "C:\...\venv_warp\lib\site-packages\warp\tests\test_codegen.py", line 22, in <module>
    wp.init()
  File "C:\...\venv_warp\lib\site-packages\warp\context.py", line 1509, in init
    runtime = Runtime()
  File "C:\...\venv_warp\lib\site-packages\warp\context.py", line 729, in __init__
    self.core = warp.build.load_dll(os.path.join(bin_path, warp_lib))
  File "C:\...\venv_warp\lib\site-packages\warp\build.py", line 299, in load_dll
    dll = ctypes.CDLL(dll_path, winmode=0)
  File "C:\Program Files\Python39\lib\ctypes\__init__.py", line 374, in __init__
    self._handle = _dlopen(self._name, mode)
FileNotFoundError: Could not find module 'C:\...\venv_warp\Lib\site-packages\warp\bin\warp.dll' (or one of its dependencies). Try using the full path with constructor syntax.
```

syigzaw, 2022-07-25:
I just tried this for Python 3.8.13 and Python 3.10.2, and I also get this error. Any developments on this?

supplient, 2022-11-06:
For python 3.9 win11 user: **winmode should be None**.

``` python
# warp/build.py
def load_dll(dll_path):    
    if (sys.version_info[0] > 3 or
        sys.version_info[0] == 3 and sys.version_info[1] >= 8):
        # dll = ctypes.CDLL(dll_path, winmode=0)   <---------- HERE
        dll = ctypes.CDLL(dll_path)
    else:
        dll = ctypes.CDLL(dll_path)
    return dll
```

I'm not sure why this works... It seems ctype has fixed the bug in python 3.9.0 ?