NVIDIA/warp #26: Remove `import *` from package

cadop, 2022-05-17:
The use of `import *` is helpful in some cases, but it makes the codebase extremely difficult to follow and debug.  For example, the recent issue for urdf https://github.com/NVIDIA/warp/issues/5#issuecomment-1126959626 seems to be a problem with the Mesh class. When following the trace in a local copy of the package, it turns out it calls the `Mesh` class.  But it seems like it should call the `Mesh` class... obviously this is confusing...

[image]

I only found the problem because after wondering why the next error (after fixing the previous one) results in `'Mesh' object has no attribute 'finalize'` , even though its in the same file and clearly has that function. 
[image]

I suspect `import *` is causing a silent problem with import resolution (not sure why the tests don't see this).

mmacklin, 2022-05-30:
Hi @cadop , I just pushed a change that cleans up some of these imports: https://github.com/NVIDIA/warp/blob/main/warp/__init__.py, can you let me know if this fixes the issue for you?

mmacklin, 2022-06-16:
Fixed in https://github.com/NVIDIA/warp/blob/6cb14da202789c230ca7d8596ecac034be5553db/warp/__init__.py

closed 2022-06-16 by mmacklin