NVIDIA/warp #28: Can install

lyd405121, 2022-05-23:


lyd405121, 2022-05-23:
- I clone this resposry and build it
- Run some example and get some usd files
[image]
- But I can not let reproduce the whole simulation but the fisrt frame
- So I guess maybe I should install warp extesion in omniverse kit
[image]

- Unfortunately， I can't find it
- The version for omniverse kit is 103.1.1 the latest

lyd405121, 2022-05-23:
- How silly I am!
- In animation layer, I found it

[image]

- But I still can not find warp extesion in omniverse kit
- Am I making another mistake?

cadop, 2022-05-23:
You need to add the warp directory to kit mini extensions. 
Check the section Push it to Git, it has an image showing what it looks like to add the directory. 
http://omniverse-docs-production.s3-website-us-west-1.amazonaws.com/kit-sdk/103.0/docs/guide/get_started_extensions.html

btw, this is better for a discussion question than an issue.

lyd405121, 2022-05-24:
> u need to add the warp directory to kit mini extensions

Thanks! I am on my way to explore warp

closed 2022-05-24 by lyd405121