NVIDIA/warp #34: Can warp differentiable with cloth materials?

fishfishson, 2022-06-22:
Material parameters of mesh like mu lamba are all floats type. Can I replace them with wp array and set them requires_grad?

cadop, 2022-06-27:
Maybe you could put this in discussion? Since its not an issue?

mmacklin, 2022-07-12:
Hi @fishfishson, in the latest release we moved the triangle elastic materials to an array so you can now get gradients through them (https://github.com/NVIDIA/warp/blob/main/warp/sim/model.py#L1306).

Cheers,
Miles

closed 2022-07-12 by mmacklin

fishfishson, 2022-07-19:
> Hi @fishfishson, in the latest release we moved the triangle elastic materials to an array so you can now get gradients through them (https://github.com/NVIDIA/warp/blob/main/warp/sim/model.py#L1306).
> 
> Cheers, Miles

Thx for updating!