NVIDIA/warp #38: Point Inside Mesh

irlanrobson, 2022-06-27:
Is the method implemented in WARP for testing if a point is inside a closed mesh correct? I am seeing that it checks the dot product of the difference vector between the query point and the closest point on the mesh with the triangle normal to see if is inside. However, e.g. in this paper they say that always using the triangle normal is incorrect if the closest point is on a vertex or an edge of the mesh. Paper: "Generating Signed Distance Fields From Triangle Meshes".

irlanrobson, 2022-08-03:
Closing it. My conclusion is that it currently keep things simple and approximate.

closed 2022-08-03 by irlanrobson

mmacklin, 2022-08-08:
Hi @irlanrobson ,

I actually have an update coming that improves robustness quite a bit (using a voting strategy based on mesh ray casts). It is a little slower, but considerably more accurate.

Will push in the next few days. Thanks for your patience.

Cheers,
Miles

irlanrobson, 2022-08-08:
Ok Miles, thanks for your reply.

mmacklin, 2022-09-23:
Hi @irlanrobson,

Just to follow up, in 0.4.3 I changed the inside/outside determination to use a voting strategy based on ray-casts. It is slower, but considerably more robust.

Cheers,
Miles