NVIDIA/warp #40: Joints of URDF are not rendered

cadop, 2022-06-28:
When running the examples and previewing in omniverse through the stored USD, it seemed like there were joints, but this was only by playing the simulation back.  However, there doesn't seem to be any joints

[image]

cadop, 2022-06-29:
@yizhouzhao  That is USD

mmacklin, 2022-09-21:
This is expected since we don't have a USD representation of joints, the output of the sim. is just time-sampled rigid body trajectories.

We will add support for the USD physics schema in the future that will allow this, but it won't be supported for URDF/MCJF sims.

closed 2022-09-21 by mmacklin