NVIDIA/warp #42: Support for Apple M1

xelatihy, 2022-07-07:
It would be nice to have support for the Apple M1, at least for the CPU device, to make `warp` code portable to more environments, in my case to use `warp` for teaching university courses. When I tried to run a simple test on the M1, the execution breaks with the error

`OSError: dlopen(/opt/homebrew/lib/python3.9/site-packages/warp/bin/warp.dylib, 0x0006): tried: '/opt/homebrew/lib/python3.9/site-packages/warp/bin/warp.dylib' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e'))`

I am sure sure if this is due to my installation or simply that `warp` does not get packaged with `armv8` support.

mmacklin, 2022-07-12:
Hi Fabio, great to hear from you :)

Just so I understand, are you building Warp yourself or using the PyPi packages? I run Warp on my M1 Macbook OK when installed through `pip`, but you're right, I think this is likely running through Rosetta, rather than natively.

I can look into adding an arm64 build, but if you can send me any more details it would be good to know why it's not running under Rosetta for you.

Cheers,
Miles

xelatihy, 2022-07-13:
Hi Miles,

it seems like I tracked down the issue. I have python installed from home-brew. Python is compiled to be arm64, checked with
```
> /usr/bin/file /opt/homebrew/bin/python3
/opt/homebrew/bin/python3: Mach-O 64-bit executable arm64
```

I then installed `warp` from `pip` and it seems to install an x86 version that python cannot load. Here is the last lines of the error log:

```
OSError: dlopen(/opt/homebrew/lib/python3.9/site-packages/warp/bin/warp.dylib, 0x0006): tried: '/opt/homebrew/lib/python3.9/site-packages/warp/bin/warp.dylib' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e'))
```

I suspect the issue is that an arm build is missing but I am not sure.

mmacklin, 2022-07-19:
Ah, I see - I guess it makes sense that Rosetta cannot run on individual shared libraries. I will look into adding an arm64 build to our build machines.

erwincoumans, 2022-08-04:
I've been using WARP on M1 using arm64 binaries by building from source code. Have you tried running 'python3 setup.py develop' in the root, instead of using pip?

mmacklin, 2022-09-21:
Just checking if I can close this now? It seems everything should be working on M1?

xelatihy, 2022-09-22:
It works fine installing from source but won't work from pip. I think you can close it, but maybe put a note in the readme about it to avoid confusion.

Kobe16, 2023-04-30:
Hi all, 

Did you end up just running `python3 setup.py develop`? I have tried running that command, but whenever I try to run one of the example files in the `Example` directory, it gives me the following message: 
`Traceback (most recent call last):
  File "/Users/kobeyang/enter/envs/warpEnv2/lib/python3.11/site-packages/warp/build.py", line 442, in load_dll
    dll = ctypes.CDLL(dll_path, winmode=0)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/kobeyang/enter/envs/warpEnv2/lib/python3.11/ctypes/__init__.py", line 376, in __init__
    self._handle = _dlopen(self._name, mode)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: dlopen(/Users/kobeyang/enter/envs/warpEnv2/lib/python3.11/site-packages/warp/bin/libwarp.dylib, 0x0006): tried: '/Users/kobeyang/enter/envs/warpEnv2/lib/python3.11/site-packages/warp/bin/libwarp.dylib' (no such file)`

I realized that this error looked similar to the one listed above, so I thought I might ask about it here. I apologize if this question is trivial -- I am quite new to this.

erwincoumans, 2023-05-04:
@Kobe16 did you run 
```
python3 build_lib.py
```
before ```python3 setup.py develop```?

Kobe16, 2023-05-19:
Hi @erwincoumans 
Sorry, I did not realize that. It is now working for me, thank you! 

Also, I have some other questions about running Warp on the M-series Macs, if you don't mind me asking. 
I noticed that whenever I run `$ pip install usd-core`, it gives this message: 
`ERROR: Could not find a version that satisfies the requirement usd-core (from versions: none)
ERROR: No matching distribution found for usd-core`
I am wondering how you worked around this issue? 

Also, I'm wondering what software you used to render the USD files on the M1 Mac? Blender?

Thank you and take care!

mmacklin, 2023-07-26:
Hi all, just to you let you know that we now ship universal binaries for macOS (x86 + aarch64). Version `0.10.1` on PyPi should now work correctly on M1 builds of Python.

Cheers,
Miles

closed 2023-07-26 by mmacklin

cholmquist, 2023-09-05:
At least for me, pip was not able to install on Mac M1.
Looking at the files on pypi there seems to be only x86_64 wheels?

https://pypi.org/project/warp-lang/0.10.1/#files

c0d1f1ed, 2023-09-05:
@cholmquist thanks for bringing that to our attention. The binaries in the `warp_lang-0.10.1-py3-none-macosx_10_13_x86_64.whl` are in fact universal (x86-64 + AArch64), but the filename incorrectly only indicates x86-64 support. I'll fix this.

nivetaiyer, 2023-12-01:
Does this binary work on Linux arm64 as well? I have a Jetson NVIDIA orin board and I simply can't get warp compile.