NVIDIA/warp #46: Bug with `wp.pow`

epi-morphism, 2022-08-04:
Seems to be a bug with `wp.pow` for negative values:
```
@wp.kernel
def test(tt: wp.array(dtype=wp.float32)):
  tt[0] = wp.pow(-1., 2.)

tt = wp.zeros((1,), dtype=wp.float32, device='cuda')
wp.launch(test, dim=1, inputs=[tt], device='cuda')
print(tt)
```
The above prints `[nan]` while it should print `[-1]`. Could be an issue with my graphics card who knows but if not then it's probably a bug somewhere.

loliverhennigh, 2022-08-22:
I was seeing the same issue on a different example. For me it was random and about 50% of computations produced the nans. My work around was to just not use `pow` and do the multiplications out.

mmacklin, 2022-09-21:
This is the default behavior for `powf()` on CUDA when using `--fast-math` (which is on by default) (https://forums.developer.nvidia.com/t/unsuspected-work-of-pow-function-pow-device-function-works-incorrectly-witn-negative-numbers/8089/2).

I agree though that this can be quite unexpected. I will revisit the use of `--fast-math`, we may actually add an option to turn this on/off per-module.

mmacklin, 2023-04-25:
Just following up, we disabled `--fast-math` by default. You can now enable it on a per-module basis using `wp.set_module_options({"fast_math": True})`.

Best,
Miles

closed 2023-04-25 by mmacklin