NVIDIA/warp #48: Calling ```wp.launch``` inside kernel

jotix16, 2022-08-09:
How is the user supposed to implement functionality with two levels of parallelism, for example matrix—matrix multiplication?

# Minimal example
My first thought was to implement two kernels, where one performs matrix—vector multiplication and the second one calls the first kernel for each column of the second matrix.

E.g. for matrixes and vectors of size 3:

``` python
@wp.func
def row_mult(row: wp.array(dtype=float), v: wp.array(dtype=float)):
    return row[0]*v[0] + row[1]*v[1] + row[2]*v[2]

@wp.kernel
def compute1(a: wp.array(dtype=float, ndim=2), b: wp.array(dtype=float), c: wp.array(dtype=float)):
    id = wp.tid()
    c[id] = row_mult(a[id], b)

@wp.kernel
def compute_parallel(a: wp.array(dtype=float, ndim=2), b: wp.array(dtype=float, ndim=2), c: wp.array(dtype=float, ndim=2)):
    id = wp.tid()
    wp.launch(kernel=compute1, dim=3, inputs=[a, b[id]], outputs=[c[id]], device="cuda")

# calling
dim = 15
a = wp.array(np.arange(9).reshape((3,3)), dtype=float, device="cuda")
c = wp.array(0*np.eye(dim, dtype=float), dtype=float, device="cuda")
b = wp.array(np.eye(dim, dtype=float), dtype=float, device="cuda")

wp.launch(kernel=compute_parallel, dim=dim, inputs=[a, b], outputs=[c], device="cuda")
print(c)
```
But apparently this causes ```RuntimeError: Could not find function wp.launch as a built-in or user-defined function. Note that user functions must be annotated with a @wp.func decorator to be called from a kernel.```

Any workaround for this?

# My use case
My use case is to compute Jacobins of a simulation unrolled trajectory, i.e.
- simulate for T steps and save the sub_computation_graphs between every two subsequential steps
- launch kernel that computes Jacobian for every step (according to its sub_computation_graph)

cadop, 2022-08-10:
@jotix16  Would you mind making this a discussion and explain how you did it? I would also be interested in this.

pourion, 2022-08-10:
At an API level, this is similar to vmap in jax and functorch, and very useful if it was possible... I only did this by multi-dimensional grids. It is related to being able to call a cuda kernel from another kernel that seems to be available according to this post: https://stackoverflow.com/questions/11747265/calling-a-kernel-from-a-kernel ?

jotix16, 2022-08-15:
@cadop Frankly, I was uncertain whether calling a kernel from a kernel was even possible in the first place. From pourion's answer, I see that this is feasible. Therefore, I am reopening the issue.

mmacklin, 2022-09-21:
CUDA does expose the ability to launch kernels from within other kernels, however it is currently not recommended due to a number of performance implications that would make it impractical for this kind of fine-grained parallelism.

This is something that will likely be improved in the future, when that happens we will look to expose it in Warp as well, but it is not on the current roadmap (will be after CUDA 12). I will close as not planned and we will revisit when new dynamic parallelism feature become available.

closed 2022-09-21 by mmacklin