NVIDIA/warp #49: Computing jacobians

jotix16, 2022-08-09:
What is the best way to compute the Jacobian  $\mathbf{J} \in \mathbb{R}^{m,n}$  of the function $\mathbf{f}(\mathbf x): \mathbb{R^n \to \mathbb{R}^m}$?

half-potato, 2022-08-27:
Manually, with a custom kernel.

mmacklin, 2022-09-21:
The Warp backwards pass computes `J^T*e` where `e` is a vector, by setting `e` to a one-hot array e.g.: `[1, 0, 0, 0...]` you can extract rows of the Jacobian, one per-backward pass.

If you have a number of environments you can also extract rows for each environment simultaneously by doing e.g.: `[1, 0, 0, 0, 1, 0, 0, 0, 1, 0, 0, 0, ...]`, where each environment has e.g.: 4 DOFs.

We will add an example and some docs on this soon.

mmacklin, 2022-09-21:
@eric-heiden for visibility.

mmacklin, 2022-09-22:
Eric just added a nice example showing how to do this for single and multiple simulations / environments: 9c08291aa5a645f1e2ba38734c3fba85dad8d0cb.

Also see new section in the docs on this approach.

Cheers,
Miles

closed 2022-09-22 by mmacklin