NVIDIA/warp #50: Any plans on adding collision handling between rigid-bodies?

jotix16, 2022-08-10:


mmacklin, 2022-09-21:
Hi @jotix16, yes, @eric-heiden is actively working on improving the rigid body simulation and should have an update soon.

The next release will also expose built-in BVH objects that will make it possible to create your own collision detection methods as well.

jotix16, 2022-09-26:
That is wonderful news. 

Thank you for letting me know your prospects, this way we avoid doing the same work twice.

Looking forward to the next release. 
Cheers!

zoezhou1999, 2023-03-30:
Hi @mmacklin @eric-heiden 
Thank you for your amazing work!
Could i know when this feature will come out? 🥺 Thank you!

eric-heiden, 2023-04-25:
Hi @jotix16, @zoezhou1999, we now have support for rigid body contacts between spheres, capsules, boxes, planes, and arbitrary meshes.
You can check out this example: https://github.com/NVIDIA/warp/blob/main/examples/example_sim_rigid_contact.py

closed 2023-04-25 by eric-heiden