NVIDIA/warp #76: any plan on supporting float64 computations?

liminchen, 2023-02-14:


kmarchais, 2023-02-14:
Hello,
With the 0.7.0 released yesterday, it is now possible to use wp.vec3d to use double precision (float64) vec3 and wp.float64 to explicitly convert a float into float64.

zeshunzong, 2023-02-14:
Hi,

Just to clarify, according to the documentation, 

> Currently Warp treats int32 and float32 as the two basic scalar compute types, and all other types as storage types. Storage types can be loaded and stored to arrays, but not participate in arithmetic operations directly. Users should cast storage types to a compute type (int or float) to perform computations.

Are we now able to use float64 for computing or is it still for storage only?

kmarchais, 2023-02-14:
Yes, I've tested and it works well. It seemed to be a bit slower though.

zeshunzong, 2023-02-15:
Hi,

Thanks for confirming that. I met some issue in installing the latest 0.7.0 version. The installed directory seems different from the previous "pip install warp-lang" so when running I got "Failed to load shared library 'home/username/.local/lib/python3.8/site-packages/warp/bin/warp.so' "
Do you have plans to update the new version to pypi?

kmarchais, 2023-02-16:
It looks like 0.7.2 is now available with pypi since today!

mmacklin, 2023-02-16:
Yes, just came here to say that it's now available :)

I will close, but please re-open if you have any issues.

closed 2023-02-16 by mmacklin

liminchen, 2023-02-16:
Thank you so much!