NVIDIA/warp #84: String arguments to a kernel launch

nithinsomu, 2023-03-21:
Is it possible to pass string type arguments as inputs in wp.launch()? Warp throws an error when I try to do it.
"RuntimeError: Error launching kernel, unable to pack kernel parameter type <class 'str'> for param bc, expected <class 'str'>"

mmacklin, 2023-04-25:
Thanks @nithinsomu, unfortunately this is expected at present.

I will add a task for us to address this for the next version.

Best,
Miles

shi-eric, 2025-01-28:
This feature fell off the radar, but if @nithinsomu or anyone else is interested in the feature, we would appreciate clarification on what is expected to be done with the string inside the kernel. Are you planning to do string comparisons, print the string, etc.?