NVIDIA/warp #85: Example with garment simulation

maria-korosteleva, 2023-03-27:
Hi! Thank you for openly sharing a simulation library, it's so valuable!

Your Readme has a video example of a full garment simulation. Is the code for this example available or is it possible to share it?
Provided cloth example is very simple 😅

mmacklin, 2023-04-25:
Hi Maria, we are looking at the possibility of releasing this asset as a sample for future Create version if licensing allows.

I will update this thread when I have confirmation.

Best,
Miles

yizhouzhao, 2023-04-29:
Thanks a lot, I am also looking for the garment demo

maria-korosteleva, 2023-05-03:
@mmacklin Thank you! The most interesting part for me is the simulator setup (self-collisions, fabric properties), and an estimate of compute time taken. 
I'm happy to provide (simple) garment meshes released under a CC4.0 license if you would like to have them:
https://zenodo.org/record/5267549#.ZFIFvHYza5d

qingximen, 2024-12-07:
Hi, do we have some updates for this question? I am also considering using Warp for garment simulation. Thanks!