NVIDIA/warp #90: fileno

mingno, 2023-04-23:
in MAYA

import warp
# Error: AttributeError: file C:\Users\mingn\AppData\Roaming\Python\Python37\site-packages\warp\__init__.py line 9: 'maya.Output' object has no attribute 'fileno' #

in Blender

>>> import warp
Traceback (most recent call last):
  File "<blender_console>", line 1, in <module>
  File "C:\Program Files\Blender Foundation\Blender 3.5\3.5\python\lib\site-packages\warp\__init__.py", line 9, in <module>
    faulthandler.enable()
io.UnsupportedOperation: fileno

mmacklin, 2023-04-25:
Hmm, thanks @mingno - this could be due to the recent addition of the `faulthandler`. I will look to disable this or at least handle these exceptions.

Best,
Miles

c0d1f1ed, 2023-04-25:
Thanks @mingno for reporting this! Could you give https://github.com/NVIDIA/warp/pull/92 a try?

mmacklin, 2023-07-26:
We have disabled the `faulthandler` on init so I will close this for now.

closed 2023-07-26 by mmacklin