NVIDIA/warp #91: Running a warp kernel from jax.jit

marcelroed, 2023-04-23:
Hi! I'm using Warp as a part of a project that uses JIT-compiled jax functions. How could I integrate Warp in this context?

Example:

```python
@jax.jit
def function(x):
    y = x + 1
    wp.launch(...)  # Modify y with a warp kernel
    return y
```

I'm looking for something like what can be done with Triton in the example in this README: https://github.com/jax-ml/jax-triton

mmacklin, 2023-04-25:
Hi @marcelroed, good question, I think this should be possible. Let me ask our JAX experts to see if they can offer some advice.

Best,
Miles

nouiz, 2023-04-28:
I don't think it is trivial. But it could be mechanical.
That part of the code isn't simple.
You can base yourself on the jax-triton code.
This jax custom operation should help you: https://jax.readthedocs.io/en/latest/Custom_Operation_for_GPUs.html

Mostly, we need to dynamically create one such custom operation.
I can help answer questions is you work on it. But don't expect this to be trivial.
I can also find someone if there is questions on the warp side of thing.

erwincoumans, 2023-05-04:
@nouiz , @marcelroed would it help to call precompiled Warp kernels from C++ like this?
https://github.com/erwincoumans/warp_cpp

marcelroed, 2023-05-04:
@erwincoumans, that looks like it'll be great for my use-case for sure, thank you! I would like to try to get this supported entirely from Python, however. Will be looking into contributing it this weekend.

erwincoumans, 2023-05-04:
> @erwincoumans, that looks like it'll be great for my use-case for sure, thank you! I would like to try to get this supported entirely from Python, however. Will be looking into contributing it this weekend.

You could write PyBind11 Python bindings for it?

tfsingh, 2023-05-06:
Following

nvlukasz, 2024-03-05:
Good news:  We have a working implementation of this that doesn't require fiddling with any native bits.  The basic usage looks like this:
```python
import warp as wp
import jax
import jax.numpy as jp

# secret sauce
from warp.jax_experimental import jax_kernel

@wp.kernel
def triple_kernel(input: wp.array(dtype=float), output: wp.array(dtype=float)):
    tid = wp.tid()
    output[tid] = 3.0 * input[tid]

wp.init()

# create a Jax primitive from a Warp kernel
jax_triple = jax_kernel(triple_kernel)

# use the Warp kernel in a Jax jitted function
@jax.jit
def f():
    x = jp.arange(0, 64, dtype=jp.float32)
    return jax_triple(x)

print(f())
```

The feature is still experimental and there are some limitations, but we plan to release it shortly because it can already be quite useful.