NVIDIA/warp #93: Omniverse Mandelbrot example crashes GPU

johndoe888, 2023-05-08:
When opening the Mandelbrot example scene in Omniverse Create or Code, the scene loads fine showing a blue texture with nothing going on (no initial Mandelbrot graphic or such).

On hitting the Play button in Create I can can see a first view of the Mandelbrot texture and then my GPU crashes immediately.

In Omniverse Code the scene opens and immediately freezes and the texture stays gray.

All the other examples work fine including the ocean_deformer (which is great).

The only thing I see is the message with assigning incompatible texture "dynamic://foo".

Unfortunately I do not see any crash report of the GPU in those logs. Is there another file or folder to look into for more about that?

What is the problem here?

System information:
AWS EC2 instance g5.4xlarge
Ubuntu 20.04
NVIDIA driver version 530.30.02
CUDA version 12.1
Cuda compilation tools, release 12.1, V12.1.105

Software used:
Omniverse Create 2022.3.3
Omniverse Code 2022.3.3
Warp with version tag 0.8.2 and self compiled warp.so

Omniverse Create with scene loaded:
[image]

Omniverse Create showing Mandelbrot texture and crashes:
[image]

Omniverse Code freezing immediately:
[image]

Omniverse Create log file:
[Create.log](https://github.com/NVIDIA/warp/files/11418741/Create.log)

Omniverse Code log file:
[Code.log](https://github.com/NVIDIA/warp/files/11418748/Code.log)

johndoe888, 2023-05-09:
Here is a crash report for the sake of completeness: 
[GPU-crash-report.txt](https://github.com/NVIDIA/warp/files/11431668/GPU-crash-report.txt)

christophercrouzet, 2023-05-30:
Hi @johndoe888, thank you for reporting this issue and apologies for the inconvenience caused!

This specific sample is building on top of Omniverse's dynamic texture (the `dynamic://` URI that you mentioned) and we also did encounter a few stability issues there internally, one of them being that a crash similar to what you are experiencing is happening on machines with multiple GPUs, which can be workaround by starting Code or Create with the flag `--/renderer/multiGpu/enabled=false`. You seem to have been using a single GPU though, so this might be something else.

We'll report this issue to the relevant teams and will update this issue accordingly so that it can get addressed in a future release of Omniverse.

Thank you for your understanding and for your kind words regarding the ocean sample! 😊

johndoe888, 2023-07-13:
Hi @christophercrouzet! 

This issue is solved from my point of view with the update to USD Composer 2023.1 and Kit 105 and Warp 0.82 on Windows. It just works fine now and can be closed from my point of view (except for the [flickering](https://github.com/NVIDIA/warp/issues/116)).

christophercrouzet, 2023-07-16:
Hi @johndoe888, thanks for letting us know! We also did notice some improvements in Kit 105 however our internal ticket related to this issue (and to the flickering) still needs to be investigated so keep in mind that there might still be some lingering issues to uncover.

closed 2023-12-05 by shi-eric