NVIDIA/warp #95: Cuda Graph: Capturing python code

jc211, 2023-05-15:
In the docs [website](https://nvidia.github.io/warp/_build/html/modules/runtime.html), it says: 

> Note that only launch calls are recorded in the graph, any Python executed outside of the kernel code will not be recorded.

However that doesn't seem to be strictly true and I was wondering if anyone could clarify what the exceptions are. For example, looking at:
https://github.com/NVIDIA/warp/blob/643464030f78f4795857911a0263a984d1db2f99/examples/example_sim_cloth.py#L103

We can see that there is a python expression that is swapping the states at every loop. Why is this expression not ignored?

jc211, 2023-05-15:
I suppose its unrolling the loop and saving the input parameters to the kernel call.

mmacklin, 2023-05-30:
Hi @jc211, that's correct - the inputs to the kernels are saved so in this case the ping/pong arguments are correctly stored in the graph.

But replaying the graph wouldn't actually affect that Python state.

Cheers,
Miles

closed 2023-05-30 by mmacklin