NVIDIA/warp #97: About differentiability across Earp and pytorch

zoezhou1999, 2023-05-22:
Hi
Thank you for this great work!
Apologize ahead if my question is stupid.
If I combine warp diff simulation with PyTorch diff renderer using warp’s PyTorch Interoperability interface and do back propagation via usual PyTorch code, will the gradient be correctly calculated? Thank you!

zoezhou1999, 2023-05-26:
I have another question: how to pass gradient of warp to PyTorch? It is not just about convert tensor to warp array during forward pass, right? are there more examples about this?

mmacklin, 2023-05-30:
Hi @zoezhou1999 ,

That's right, you need to create a custom Torch function and implement the backwards pass (using the `warp.Tape` object).

There is an example here of how to do it for a simulation example:

https://github.com/NVIDIA/warp/blob/main/examples/example_sim_fk_grad_torch.py#L53

Let me or @eric-heiden know if you have questions. I think we should probably add an even simpler example to show how to interop Torch / Warp.

Cheers,
Miles

closed 2023-07-26 by mmacklin

TheDevilWillBeBee, 2024-02-16:
Hi,

Thanks for creating this amazing framework. Can you provide a simple example of backpropagation where most of the computation (including the loss function) happens in PyTorch and Warp is only used to implement one of the modules.

Thank you in advance!