NVIDIA/warp #113: How to produce animation with Warp

yuhan-zh, 2023-06-30:
Hello!
       I am wondering how can I produce animation like what is shown in the first gif (human with dress dancing) shown in the Readme with Warp.

mmacklin, 2023-07-02:
Hi @yuhan-zh, Warp supports collision with deforming bodies / meshes, please take a look at https://github.com/NVIDIA/warp/blob/main/examples/example_mesh.py#L126C19-L126C19 for an example of how to update an existing `wp.Mesh` object. 

For the dancing character we exported time-sampled USD vertex animation from Blender, then per-frame we update the collision geometry accordingly. Please take a look at https://github.com/NVIDIA/warp/blob/main/exts/omni.warp/omni/warp/nodes/OgnCloth.py for an example of how the Omniverse integration looks like. This also shows how to compute per-vertex velocities given the mesh at two keyframes.

Hope that helps.

Best,
Miles

yuhan-zh, 2023-07-04:
@mmacklin Thanks for your reply! The recommendations help a lot. And btw I really hope you can release the example code of the dancing character, it really looks amazing.

Best,
Yuhan

closed 2023-12-05 by shi-eric