NVIDIA/warp #114: Seg fault with SimRendererOpenGL and springs

ShaoxiongYao, 2023-06-30:
Hi,

Warp version 0.9.0, running on Ubuntu 20.04, Nvidia Driver Version: 470.182.03, CUDA Version: 11.4

I am trying the new rendering interface `SimRendererOpenGL` like the one used in [https://github.com/NVIDIA/warp/blob/main/examples/example_sim_rigid_fem.py](https://github.com/NVIDIA/warp/blob/main/examples/example_sim_rigid_fem.py). However, if there are springs in the model, the opengl render will cause segmentation fault. The following code is a copy of [https://github.com/NVIDIA/warp/blob/main/examples/example_sim_particle_chain.py](https://github.com/NVIDIA/warp/blob/main/examples/example_sim_particle_chain.py), only the render is changed.

```
import os
import math

import warp as wp
import warp.sim
import warp.sim.render

wp.init()

class Example:
    def __init__(self, stage):
        self.sim_width = 64
        self.sim_height = 32

        self.sim_fps = 60.0
        self.sim_substeps = 10
        self.sim_duration = 5.0
        self.sim_frames = int(self.sim_duration * self.sim_fps)
        self.sim_dt = (1.0 / self.sim_fps) / self.sim_substeps
        self.sim_time = 0.0

        builder = wp.sim.ModelBuilder()

        # anchor
        builder.add_particle((0.0, 1.0, 0.0), (0.0, 0.0, 0.0), 0.0)

        # chain
        for i in range(1, 10):
            radius = math.sqrt(i) * 0.2
            mass = math.pi * radius * radius * radius
            builder.add_particle((i, 1.0, 0.0), (0.0, 0.0, 0.0), mass, radius=radius)
            builder.add_spring(i - 1, i, 1.0e6, 0.0, 0)

        self.model = builder.finalize()
        self.model.ground = False

        self.integrator = wp.sim.XPBDIntegrator()

        self.state_0 = self.model.state()
        self.state_1 = self.model.state()

        # NOTE: change the renderer
        # self.renderer = wp.sim.render.SimRenderer(self.model, stage, scaling=1.0)
        self.renderer = wp.sim.render.SimRendererOpenGL(self.model, stage, scaling=1.0)

    def update(self):
        with wp.ScopedTimer("simulate"):
            for s in range(self.sim_substeps):
                self.state_0.clear_forces()
                self.state_1.clear_forces()

                self.integrator.simulate(self.model, self.state_0, self.state_1, self.sim_dt)
                self.sim_time += self.sim_dt

                # swap states
                (self.state_0, self.state_1) = (self.state_1, self.state_0)

    def render(self, is_live=False):
        with wp.ScopedTimer("render"):
            time = 0.0 if is_live else self.sim_time

            self.renderer.begin_frame(time)
            self.renderer.render(self.state_0)
            self.renderer.end_frame()

if __name__ == "__main__":
    stage_path = os.path.join(os.path.dirname(__file__), "outputs/example_sim_particle_chain.usd")

    example = Example(stage_path)

    for i in range(example.sim_frames):
        example.update()
        example.render()

    example.renderer.save()
```

Output: 
```
Warp 0.9.0 initialized:
   CUDA Toolkit: 11.5, Driver: 11.4
   Devices:
     "cpu"    | x86_64
     "cuda:0" | Quadro P4000 (sm_61)
   Kernel cache: /home/motion/.cache/warp/0.9.0
Module warp.render.render_opengl load on device 'cuda:0' took 533.40 ms
        Module warp.sim.integrator_euler load on device 'cuda:0' took 52.60 ms
        Module warp.sim.integrator_xpbd load on device 'cuda:0' took 81.54 ms
simulate took 137.37 ms
Segmentation fault (core dumped)
```

ShaoxiongYao, 2023-06-30:
It appears that the bug was caused by line 239 in `warp/sim/render.py`
[https://github.com/NVIDIA/warp/blob/0674dcef7354a23244a9d646d98454d079cb838a/warp/sim/render.py#L239C21-L239C120](url). 

Change: `self.render_line_list("springs", particle_q, self.model.spring_indices.numpy().flatten(), [], 0.05)` to `self.render_line_list("springs", particle_q, self.model.spring_indices.numpy().flatten(), None, 0.05)` fixed the problem.

mmacklin, 2023-07-02:
Thank you @ShaoxiongYao! We will integrate this fix in the next release.

ShaoxiongYao, 2023-08-06:
The problem was fixed in 0.10.1.

closed 2023-08-06 by ShaoxiongYao