NVIDIA/warp #116: Omniverse USD Composer Mandelbrot example flickering

johndoe888, 2023-07-12:
The Mandelbrot example shows some disturbing flickering when going full-screen and hovering around (and sometimes just so).

I think the dynamic textures to be supercool for displaying high-res calculated Warp results, but with the flickering very difficult to work with and show the results to others.

Environment:
AWS EC2 g5.4xlarge connected via RemoteDesktop
USD Composer 2023.1
Warp 0.82 (the standard that came with the installation of USD Composer 2023.1)

christophercrouzet, 2023-07-12:
Hi @johndoe888, thanks for sharing this issue!

The flickering that you mentioned was also reported on our end alongside the other issue that you ran into with #93. The corresponding team has assigned a high priority on it it so we're expecting it to be addressed in an upcoming release of the Composer app.

In the meantime, one thing that we observed is that reducing the scene FPS _seems_ to help with the flickering. In our case, 60 FPS seemed to behave more or less as expected.

Thank you for your patience!

johndoe888, 2023-07-13:
For me 24 FPS did the trick. 60 FPS still flickered in fullscreen mode. Maybe it is different across RemoteDesktop.

shi-eric, 2024-06-06:
@christophercrouzet Is this fixed now?

christophercrouzet, 2024-06-06:
Yes, it should have been addressed in more recent releases of USD Composer!

closed 2024-06-06 by christophercrouzet