NVIDIA/warp #120: FeatureRequest: wp.clamp that works on vectors

jc211, 2023-07-18:
Currently wp.clamp is only defined on scalars, would be good to have one on vectors too.

c0d1f1ed, 2023-07-24:
Thanks for the feature request. This sounds reasonable at first glance, but I'm a bit concerned about feature creep and bloat. There are additional types that could potentially have their own clamp function, and it's difficult to determine where to draw the line. Each of them should ideally be continuously tested, it causes maintenance overhead, and JIT-compile time would be negatively affected.

Is it reasonable to just define your own `@wp.func` functions that perform vector clamping using `wp.clamp()`, instead? Note this is also the expectation when writing CUDA code, although [helper header libraries](https://github.com/NVIDIA/cuda-samples/blob/master/Common/helper_math.h#L1168) exist.

jc211, 2023-07-24:
Thank you for considering the request. I think not including it to avoid feature creep is fair enough. Writing the function is simple enough that it doesn't need to be included I suppose.

closed 2024-07-18 by shi-eric