NVIDIA/warp #122: uint16 and uint8 inside struct do not work

jc211, 2023-07-22:
I get a "CUDA build failed" error when I include a wp.uint16 or wp.uint8. 

```
warp/warp/native/builtin.h(1211): error: no instance of overloaded function "atomicAdd" matches the argument list argument types are: (wp::uint16 *, wp::uint16)
      return atomicAdd(buf, value);
             ^
          detected during instantiation of "T wp::atomic_add(T *, T) [with T=wp::uint16]" at line 41 of default_program```
```

Example:
```python
import warp as wp

wp.init()
@wp.struct
class Foo:
    # temp: wp.uint32 # works
    temp: wp.uint16 # does not work
    # temp: wp.uint8 # does not work

@wp.kernel
def test_kernel(
    foos: wp.array(dtype=Foo)
):
    foo = Foo()
    foos[0] = foo

n = 1
foos = wp.empty((n, ), dtype=Foo)
wp.launch(
    kernel=test_kernel,
    dim=(n,),
    inputs=[
        foos
    ],
)
wp.synchronize()
```

nvlukasz, 2023-07-26:
Hi @jc211, this is actually a deeper CUDA limitation.  Atomic operations are only supported for 32-bit and 64-bit integers and floating point types.

We should catch this earlier and improve the error message at the Warp level, thanks for reporting.

nvlukasz, 2023-07-26:
Oh, I see.  That seems unrelated.  Let me dig into it.

nvlukasz, 2023-07-26:
Looks like a fairly recent regression.  I don't see an obvious workaround yet, but we're working on a fix.

c0d1f1ed, 2023-07-31:
I believe these atomics are implicitly generated for the adjoint functions. I recently had to implement 16-bit float atomic addition support for Clang/LLVM, because the existing inline assembly didn't produce the correct result. The (currently still default) use of NVRTC *should* be unaffected by that but it at least shows that these operations are tricky. The latest PTX 8.1 ISA only defines `cas.b16` and `add.f16` atomic operations; every other operation on types smaller than 32-bit must be emulated. The hardware itself can't do any 8-bit atomic operations, as far as I can tell (emulating that probably requires a CAS which accesses a neighboring byte which must remain unchanged).

I think until we can confidently support all edge cases we should produce a gentle error that doesn't allow 8- and 16-bit types in structs for adjoint functions. When the user doesn't explicitly ask for adjoint functions I believe it shouldn't produce an error but a warning? @nvlukasz I'll leave this in your hands for now but let me know if you need me to look more into emulating these operations or make changes to the compilation steps.

jc211, 2023-08-01:
Re: Implicitly generated adjoint functions - I don't know if this is related, but I did notice that when one of my warp kernels has an error, warp will complain about other kernels that have unused adjoint-related variables. This somewhat pollutes the error messages that are relevant to my bug especially considering I never asked for the adjoints  to be created anywhere.

shi-eric, 2023-12-05:
I'll close the issue since the original reported problem seems to be fixed as of `1.0.0b2`, which is available on PyPI.

closed 2023-12-05 by shi-eric