NVIDIA/warp #124: Directly changing struct element inside kernel does not work

jc211, 2023-07-25:
A wp.kernel cannot directly change a single element within a struct. 

Example:
```python 
import warp as wp

@wp.struct
class Foo:
    temp: wp.uint32

@wp.kernel
def test_kernel(
    foos: wp.array(dtype=Foo)
):
    i = wp.tid()
    foos[i].temp = wp.uint32(1) # does not work

    # foo = foos[i]
    # foo.temp = wp.uint32(1) # works
    # foos[i] = foo

wp.init()
foos = wp.zeros((10, ), dtype=Foo)
print(foos)
wp.launch(
    kernel=test_kernel,
    dim=(10,),
    inputs=[
        foos
    ],
)
wp.synchronize()
print(foos) # foos is still all zero
```

c0d1f1ed, 2023-07-31:
This is on our radar as a very desirable feature! I'll implement it after taking care of some slightly higher priority work.

c0d1f1ed, 2023-11-24:
This feature is now supported: https://github.com/NVIDIA/warp/blob/79a56a9c2754bbec1418e490efafd43091056856/warp/tests/test_lvalue.py#L256

closed 2023-12-05 by c0d1f1ed