NVIDIA/warp #125: Question on tape

jc211, 2023-07-25:
"Warp uses a source-code transformation approach to auto-differentiation."

I don't quite understand what this means and I know this is not how one is supposed to use tape but my question is: Is it technically possible for me to write a ```@wp.func foo``` and get back a ```@wp.func foo_derivative``` which I can save and reuse later within kernel launches? This is in contrast to doing it at the kernel level.

mmacklin, 2023-07-26:
Hi @jc211,

Currently we don't provide a way to call the derivative directly - however this would not be too difficult and I can imagine could be quite useful. I will make a note of this feature request.

Thanks,
Miles

jsfrancal, 2025-09-13:
Dear all, what is the status of this issue? I have a use case where calling the automatically generated adjoints of user functions would be quite useful (to avoid having to re-code the whole adjoint chain for a customized adjoint. Given that the adjoint functions of user functions are already generated, I expect this request to be quite easy to implement, provided that a sensible naming convention is chosen. Thanks.

daedalus5, 2025-12-30:
@jsfrancal @jc211  this feature was added in commit [f12e243](https://github.com/NVIDIA/warp/commit/f12e243a0ccabc8c04e9d98e53733380ee8c1199)

We added `warp.grad()` to the API. Full documentation will be included with the 1.11 release (should be end of this week). You can also rebuild the docs from main to get a sneak preview.