NVIDIA/warp #127: OpenGLRenderer: render_points radius cannot be wp.array

jc211, 2023-07-26:
In render_points, radius still cannot be a wp.array. I think it should be allowed since the sim model stored its radii as wp.array.

shi-eric, 2025-03-10:
@christophercrouzet to provide more information about why this won't be done.

christophercrouzet, 2025-03-16:
Hi @jc211,

Our current intention for the OpenGL renderer is to keep it simple for quick debugging purposes, rather than supporting all the features typically expected from traditional renderers. See our recently updated documentation on the topic [here](https://nvidia.github.io/warp/modules/render.html#standalone-renderers).

Therefore, we encourage you to use the USD renderer instead, which offers better support for these features. You can then load the resulting file into other programs, such as Omniverse Kit or Blender, to generate higher quality renders.

Apologies for the _very_ delayed and possibly disappointing reply.

closed 2025-03-16 by christophercrouzet