NVIDIA/warp #129: Indexing into wp.array of 2 dimensions

jc211, 2023-07-28:
Is it possible to support indexing into a 2 dimensional array with either 2 numbers or the index? At the moment, an error is raised if a 2d array is accessed with just a 1D index.

mmacklin, 2023-07-31:
Hi @jc211, I'm not sure exactly what you meant by indexing with 1D index gives an error, here's an example: 
```

import warp as wp
wp.init()

@wp.kernel
def compute2d(a: wp.array2d(dtype=float)):

    i, j = wp.tid()

    a[i, j] = float(i)*10.0 + float(j)
    

a = wp.zeros((10,10), dtype=float)

@wp.kernel
def compute1d(a: wp.array2d(dtype=float)):

    i, j = wp.tid()

    row = a[i]

    # print first entry of each row
    print(row[0])
    

a = wp.zeros((10,10), dtype=float)

wp.launch(compute2d, dim=[10, 10], inputs=[a])
wp.launch(compute1d, dim=[10], inputs=[a])

print(a)
```

The first kernel fills a 2D array with unique values (using 2 indices), the second kernel prints out the first entry in each row (using 1D indexing into the 2D array to get the row).

Does that help?

jc211, 2023-08-01:
Yes thank you. I thought that indexing into a 2d array with only one number eg array2d[100] would access the 100th element in memory. But I see that warp gives the 100th row instead.

closed 2023-08-01 by jc211