NVIDIA/warp #130: How can I get bvh identifier and use relevent kernel?

ZiXuanVickyLu, 2023-07-29:
In the document, I found a kernel called warp.bvh_query_aabb(), and the first parameter is a bvh identifier. But I do not find there is a 'bvh' type (which should exist a builder, I guess?) where I can refer it with a identifier. In the document, it indicates that bvh is associated with a Mesh, I guess?
So how can I use this kernel? Or where can I get the bvh identifier?
(I sincerely ask for a simple example :-)
[image]

mmacklin, 2023-07-31:
Hi @ZiXuanVickyLu, please find an example of using the Bvh class here: https://github.com/NVIDIA/warp/blob/main/warp/tests/test_bvh.py#L81.

We will work on better documentation for this feature as well.

Cheers,
Miles

closed 2023-07-31 by mmacklin

ZiXuanVickyLu, 2023-07-31:
Appreciate!