NVIDIA/warp #131: Clamp particle velocity to a vmax that assures system stability

NRHuntoon, 2023-07-31:
Semi-Implicit integrators benefit from having maximum velocities to ensure stability. Numerical stability of `SemiImplicitIntegegrator` would be enhanced if a maximum velocity was passed into the kernel and any particle velocity that exceeded this was clamped.

Passing either a global vmax based on the minimum mesh dimensions, or a per particle vmax array would be ideal.

mmacklin, 2023-08-02:
Thanks @NRHuntoon, as discussed on Discord this makes sense to me - @eric-heiden could you take a look at implementing this?

eric-heiden, 2023-08-24:
@NRHuntoon we will have an option for a global maximum on particle velocity in the next release which will clamp the velocity magnitude.

shi-eric, 2023-12-05:
`Model.particle_max_velocity` added in 1.0.0-beta.1.

closed 2023-12-05 by shi-eric