NVIDIA/warp #134: warp.sim.state flatten method hields `AttributeError: module 'warp' has no attribute 'is_tensor'`

rcremese, 2023-08-01:
Hello, 
I'm new with wrap and already amazed by what it can achieve.
While I was trying the simulation module with warp V0.10.1 I ran into the following issue

Using the code snippet provided in the [documentation](https://nvidia.github.io/warp/_build/html/modules/sim.html#warp.sim.ModelBuilder) I tried to print the output of the `state.flatten()` method. The error I get is :
```shell
File "/home/rcremese/mambaforge/envs/warp-env/lib/python3.10/site-packages/warp/sim/model.py", line 216, in flatten
    if wp.is_tensor(value):
AttributeError: module 'warp' has no attribute 'is_tensor
```

To reproduce here is the code I used : 
```python
# import warp
import warp as wp
import warp.sim

wp.init()
builder = warp.sim.ModelBuilder()

# anchor point (zero mass)
builder.add_particle((0, 1.0, 0.0), (0.0, 0.0, 0.0), 0.0)

# build chain
for i in range(1,10):
    builder.add_particle((i, 1.0, 0.0), (0.0, 0.0, 0.0), 1.0)
    builder.add_spring(i-1, i, 1.e+3, 0.0, 0)

# create model
model = builder.finalize("cuda")

state = model.state()
integrator = warp.sim.SemiImplicitIntegrator()

for i in range(100):
   print(state.flatten())
   state.clear_forces()
   integrator.simulate(model, state, state, dt=1.0/60.0)
```

I know the module is in active development but I wanted to point out this issue

mmacklin, 2023-08-02:
Huh, this looks like a hangover from some long gone PyTorch code. For now I would avoid the method, although you could likely replace this particular call to `isinstance(value, wp.array)` to gather a list of all arrays on the `State` object. This may not be super useful though, since there are generally not too many attributes so you can generally enumerate them manually.

We used to use this flatten() method to gather a list of all PyTorch tensors for gradient tracking. We will likely remove this method in the next version, thanks for bringing this to our attention!

rcremese, 2023-08-04:
I won't use this method as access to the state attributes is straightforward but I thought it would be usefull to know the bug.

shi-eric, 2023-12-05:
Closing because this has been fixed as of `1.0.0b2`, which is available on PyPI. Using `Model.flatten()` also currently prints the following deprecation notice:
```
DeprecationWarning: Model.flatten() will be removed in a future Warp version.
```

closed 2023-12-05 by shi-eric