NVIDIA/warp #136: Is there a way to get all requires_grad=True variable and do gradient clipping inside simulator of warp？

zoezhou1999, 2023-08-06:
Hi
Is there a way to get all requires_grad=True variable and do gradient clipping inside simulator of warp？Thank you!

mmacklin, 2023-08-17:
Hmm.. that sounds useful, are you using the `wp.Tape()`? That keeps track of all the gradients used during the computation. You could add a post process to clip the gradients there I think?

zoezhou1999, 2023-08-17:
Hi Miles
Thank you for the reply. I will look at it. Btw, when I played warp and pytorch, I found warp cannot support pytorch's retain_graph=True functionality, i.e., loss.backward(retain_graph=True). But not using retain_graph solves the issue.

Leon-LXA, 2024-02-05:
> Hmm.. that sounds useful, are you using the `wp.Tape()`? That keeps track of all the gradients used during the computation. You could add a post process to clip the gradients there I think?

When I try to change the value of gradients inside Tape function, it showed that 'operation would make the legacy stream depend on a capturing blocking stream'. Seems like I cannot do any modification of gradients' value.