NVIDIA/warp #137: Actuation model like Mujoco

areiner222, 2023-08-09:
Hi,

Is it possible to apply forces to a gear like actuation model for a rigid body Model simulation? I'd like to actuate gears at joints similarly to how you can do it via ctrl in Mujoco. I noticed the actuators are included in the nv_humanoid.xml example, but I'm not sure how they are used.

Thanks for your help!

mmacklin, 2023-08-17:
@eric-heiden should be able to help.

eric-heiden, 2023-08-22:
Hi @areiner222,

We currently do not have support for the actuation models as they are in Mujoco. At the moment, all joint torques (all controllable degrees of freedom, irrespective of the definition in the Mujoco file) can be set in `Model.joint_act` which applies these controls defined in generalized coordinates as body forces directly. You could implement you own actuation model based on this, or directly work in maximal coordinates by applying linear and angular forces to the `state.body_f` array at every simulation step.

closed 2025-01-03 by shi-eric