NVIDIA/warp #140: Tape back propagation breaks when used with loops of size which are calculated inside the function

nithinsomu, 2023-08-16:
I am trying to back propagate and find the gradients of a variable whose value is defined by a looped calculation where the size of loop depends on the parameter passed. Tape backwards seems to return zero as the gradient. When I manually fix the size of the loop, tape produces the right gradient value. The code can be found below : 
 
import warp as wp
import numpy as np
wp.init()

@wp.func
def do_sum(a : wp.array(dtype=float),
                weight : wp.array(dtype=float),
                c : wp.array(dtype=float),
                p : int):
    summ=float(0.0)
    
    
    for j in range(p+1):
        summ = summ + weight[j] * a[j] 
        
    summ = summ * c[0]
    return summ
    
@wp.kernel
def calculations(a : wp.array(dtype=float),
                weight : wp.array(dtype=float),
                c : wp.array(dtype=float),
                summ : wp.array(dtype=float),
                p : int):
    i = wp.tid()
    summ[i] = do_sum(a, weight, c, p)
    
a = wp.array(np.array([1,2,3,4,5]), dtype=float)
weight = wp.array(np.array([1,2,3,4,5]), dtype=float)
c = wp.array(np.array([1]), dtype=float, requires_grad=True)
summ = wp.zeros(1, dtype=float, requires_grad=True)
p=4
tape = wp.Tape()
with tape:
    wp.launch(calculations, dim=1, inputs=[a, weight, c, summ, p])
    
tape.backward(summ)

c.grad.numpy()

The loop is inside the function "do_summ". In this code I am setting the size of loop as p+1 and for which tape provides zero as the gradient value of 'summ' with respect to variable 'c'. If we fix the loop size to be 5 (based on the size of the array used in this case), tape provides the right value of gradient "55".
Kindly provide some insights on this.  Not sure if I am committing any mistakes in the calculation or if such variable sized loop operations aren't supported in tape operations.  Thanks in advance.

mmacklin, 2023-08-17:
That's right, the contents of dynamic loops are not replayed in the backwards pass. That means anything they compute needs to be stored out to an array somewhere. Since you are actually storing the sum out a simple way to re-write this is as follows, where I have moved the scaling factor out to the calling function (so it operates on the value coming from the array, rather than the value computed during the loop replay).

```import warp as wp
import numpy as np
wp.init()

@wp.func
def do_sum(a : wp.array(dtype=float),
	weight : wp.array(dtype=float),
	p : int):
	summ=float(0.0)

	for j in range(p+1):
		summ = summ + weight[j] * a[j] 
		
	summ = summ
	return summ

@wp.kernel
def calculations(a : wp.array(dtype=float),
	weight : wp.array(dtype=float),
	c : wp.array(dtype=float),
	summ : wp.array(dtype=float),
	p : int):

	i = wp.tid()
	summ[i] = do_sum(a, weight, p)*c[0]

a = wp.array([1,2,3,4,5], dtype=float, requires_grad=True)
weight = wp.array([1,2,3,4,5], dtype=float, requires_grad=True)
c = wp.array(np.array([1]), dtype=float, requires_grad=True)
summ = wp.zeros(1, dtype=float, requires_grad=True)
p=4
tape = wp.Tape()

with tape:
	wp.launch(calculations, dim=1, inputs=[a, weight, c, summ, p])

tape.backward(summ)

print(c.grad.numpy())
```

This stuff is fairly subtle and can be quite confusing in general - I think we could do a better job at describing the rules for correct auto-diff but hope that helps.

Best,
Miles

closed 2023-08-17 by mmacklin

c0d1f1ed, 2023-08-17:
Setting a fixed loop iteration count of 5 will cause the loop to be unrolled, which in turn allows for automatic differentiation used by the backward tape execution. We have no firm plans for supporting differentation of loops with variable iteration count at this point.

That said I concur that we could improve documentation around this, as well as provide clearer warnings and advise on workarounds. I'll try to get around to that.

nithinsomu, 2023-08-17:
Thank you guys for looking into this. Really appreciate it.
Thanks,
Nithin