NVIDIA/warp #144: Unstable integration with Symplectic Euler

ClemensSchwarke, 2023-08-24:
Hi,

I am trying to run the quadruped example using the provided `SemiImplicitIntegrator` but the simulation is highly unstable. I tried to tune the stiffness parameters but with no success. In my understanding, this behavior is mainly caused by the constraint forces between joints. Is there anything I might be overlooking?

Thanks in advance!

eric-heiden, 2023-08-24:
Hi Clemens,

The Euler integrator needs a finer time resolution to remain stable. So you can try a higher number of `sim_substeps` here (e.g. 16 worked for me):
https://github.com/NVIDIA/warp/blob/main/examples/example_sim_quadruped.py#L35

ClemensSchwarke, 2023-08-25:
Thanks a lot for the quick and helpful response Eric! And sorry for missing the obvious.

Cheers,
Clemens

closed 2023-08-25 by ClemensSchwarke