NVIDIA/warp #150: Adding cubic root

Dolorousrtur, 2023-10-24:
Hey, thanks to the developers for this incredible library!

In my project, I need to precisely compute the cubic roots of double-precision variables. But it seems that warp does not have it implemented (warp.pow(x, 1/3) is not precise enough.

I reckon it shouldn't be too difficult to add it since it only needs to call `cbrf` function in CUDA. I tried to find a way to add cbrf to the source code of Warp same as it is done with `sqrt`, but couldn't do that yet.

Could you direct me to how (in which files) the `sqrt` support is implemented, s.t. I cat add `cbrf` the same way?

Many thanks in advance!

daedalus5, 2023-10-25:
Hi @Dolorousrtur, thanks for your request. `cbrt` will be in the next Warp release.

Dolorousrtur, 2023-10-27:
Hi, @daedalus5 ! That's great!

In the meantime, could you please describe the steps I can follow to add it to the code myself s.t. I can build the updated version from sources? If it won't take too much time for you.

Thank you in advance!

shi-eric, 2024-01-16:
`cbrt` was included in the `v1.0.0-beta.5` release: https://github.com/NVIDIA/warp/releases

closed 2024-01-16 by shi-eric