NVIDIA/warp #151: SLANG.D as differentiable renderer for Warp

korzen, 2023-10-26:
Hello,

I have just come across [https://research.nvidia.com/labs/rtr/publication/bangaru2023slangd/](url) and it instantly got me thinking whether SLANG.D could be used as a lightweight differentiable rasterizer or raytracer (RT-Cores accelerated?) for simulation in Warp? Some example integration would be great. Cheers!

mmacklin, 2023-11-20:
Hi @korzen, agreed, I think that it can totally make sense to do diff. render in something like SLANG.D and physical simulation in Warp. We do have an example of a differentiable raytracer in Warp here: https://github.com/NVIDIA/warp/blob/main/examples/example_diffray.py, but will look into how we might integrate with SLANG as well.

Best,
Miles

jprnd, 2024-01-23:
could this be done with the Kaolin Library [https://kaolin.readthedocs.io/](https://kaolin.readthedocs.io/)? Are there advantages to SLANG.D over Kaolin as the diff renderer?@mmacklin any advice on interop? Can I just keep everything in USD and stream from warp to Kaolin?