NVIDIA/warp #153: Failed to build CPU module because no CPU buildchain was found

Dolorousrtur, 2023-10-27:
Hi,

I'm trying to build the warp library from sources with the commands
```
python build_lib.py
pip install -e .
```

It builds successfully, but when running `wp.launch` it yields the following error:

```
File ~/Workdir/00_Projects/warp/warp/context.py:3358, in launch(kernel, dim, inputs, outputs, adj_inputs, adj_outputs, device, stream, adjoint, record_tape, record_cmd)
   3356 # delay load modules, including new overload if needed
   3357 module = kernel.module
-> 3358 if not module.load(device):
   3359     return
   3361 # late bind

File ~/Workdir/00_Projects/warp/warp/context.py:1364, in Module.load(self, device)
   1362         return False
   1363     if not warp.is_cpu_available():
-> 1364         raise RuntimeError("Failed to build CPU module because no CPU buildchain was found")
   1365 else:
   1366     # check if already loaded
   1367     if device.context in self.cuda_modules:

RuntimeError: Failed to build CPU module because no CPU buildchain was found
```

I'm running Ubuntu 22.04 with C++ compiler installed via conda.

When installing warp from pip (`pip install warp-lang`) everything works fine.

Also, when building from sources with CUDA support (`python build_lib.py -cuda_path=*CUDA_PATH*`) it runs fine on GPU, but yields the same error on CPU.

Could you please guide me on how this problem can be solved? Thank you in advance!

c0d1f1ed, 2023-10-28:
Thanks for reporting this. `build_lib.py` should build a "standalone" CPU buildchain and produce the `warp/bin/warp-clang.so` library. If the build failed and it's not there, or the library fails to be loaded, `warp.is_cpu_available()` will return `False` and that error will be produced when attempting to run Warp kernels on the CPU.

One scenario under which we've observed issues before is when building Warp inside an environment with a recent C++ compiler version and then attempting to run Warp from an environment with an older C++ runtime version. This can produce errors such as:
```
OSError:
/.../libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /.../warp/warp/bin/warp-clang.so)
```
But it sounds like you didn't encounter any errors prior to the "no CPU buildchain" one?

Would you be able to check what happens at the `self.llvm = self.load_dll(llvm_lib)` line in `warp/context.py`?

c0d1f1ed, 2023-11-27:
@Dolorousrtur do you have any update on this? Does the issue still occur with the latest version of Warp or did you figure out the problem?

closed 2023-12-05 by c0d1f1ed