NVIDIA/warp #154: Custom Gradient Function Variable Naming

ClemensSchwarke, 2023-10-31:
Hi,
When I create a custom gradient function like:

`@wp.func_grad(example)
def adj_example(a: float, b: float, adj_ret: float):

	...
	
    if (a < 0.0):
        adj_a = adj_ret

    wp.adjoint[a] += adj_a`

the generated code looks like this:

`static CUDA_CALLABLE void adj_example(
    float32 var_a,
    float32 var_b,
    float32 & var_adj_ret)
{

	...

    // if (vn < 0.0):                                                                         <L 29>
    var_9 = (var_a < var_0);
    if (var_9) {
        // adj_a = adj_ret                                                                     <L 30>
        wp::copy(var_10, var_7);
    }
    // wp.adjoint[a] += adj_a
	var_11 = wp::add(adj_a, adj_a);
    adj_a = var_11;
}`

First, adj_a is substituted with var_10 in the `wp::copy` call. In the next appearance of adj_a in `wp::add`, it is not substituted with var_10 anymore. This is very unexpected (at least for me). I would be happy for any insights on this!

Thanks, 
Clemens

eric-heiden, 2023-11-06:
Hi Clemens,

This is definitely a bug where the variable names seem to clash with the `adj_a` variable. We will have this fixed in the next release, but for now you could try a different variable name and clear the cache.

```py
@wp.func_grad(example)
def adj_example(a: float, b: float, adj_ret: float):
    # ...

    temp = 0.0  # use a variable name other than `adj_a` here and initialize with 0
    if a < 0.0:
        temp = adj_ret

    wp.adjoint[a] += temp
```

ClemensSchwarke, 2023-11-07:
Hi Eric,
thanks for your suggestion, it is indeed easy to fix so not an issue for me. Just hard to identify in the first place, so thanks for looking into it!

closed 2023-11-07 by ClemensSchwarke