NVIDIA/warp #160: update the mesh during simulation

cyanzhao42, 2023-11-29:
Hi, thanks for the great work. I am interested in simulating cloth with moving objects. I am following the example code: https://github.com/NVIDIA/warp/blob/main/examples/example_sim_cloth.py here. I am wondering if it would be possible to update the "mesh"  (builder.add_shape_mesh) vertices during the simulation.

I found the example here - https://github.com/NVIDIA/warp/blob/main/examples/example_mesh.py, on how to update the mesh, however, the mesh object is not compatible with wp.sim. I am wondering if there is a smart way to address this. Thanks!

mmacklin, 2023-12-12:
Hi @cyanzhao42, yes take a look at the `wp.sim.Mesh` class: https://github.com/NVIDIA/warp/blob/main/warp/sim/model.py#L256. Here you can see it contains a regular mesh that can be updated + refit() as desired.

Cheers,
Miles

mmacklin, 2023-12-12:
The Omniverse cloth node extension also does this, you can look here for an example:

https://github.com/NVIDIA/warp/blob/main/exts/omni.warp/omni/warp/nodes/_impl/OgnClothSimulate.py#L368

Cheers,
Miles

closed 2024-06-06 by shi-eric

SangHunHan92, 2024-07-04:
Thanks for great work !

I want to simulate a cloth for a moving object, exactly like this question.
But I couldn't find updated + refit() in the link below:
https://github.com/NVIDIA/warp/blob/main/warp/sim/model.py#L256

Has the function to update vertices been moved? So how is it available now?

SangHunHan92, 2024-07-04:
https://github.com/NVIDIA/warp/blob/bfbd0258479ea1378788c2ff5bdbbdc1da385612/warp/types.py#L2997-L3004
https://github.com/NVIDIA/warp/blob/bfbd0258479ea1378788c2ff5bdbbdc1da385612/warp/examples/core/example_mesh.py#L116-L129

Is this what you were referring to?