NVIDIA/warp #164: RuntimeError: CUDA is not available

mahdian-organization, 2023-12-02:
Hi there, 
I got an error when I ran my code. My graphics card is a `GeForce GT 730M`.
```
Warp CUDA error: Failed to get function cuGetProcAddress
Warp 0.10.1 initialized:
   CUDA devices not available
   Devices:
     "cpu"    | Intel64 Family 6 Model 60 Stepping 3, GenuineIntel
   Kernel cache: C:\Users\a.mahdianrad\AppData\Local\NVIDIA Corporation\warp\Cache\0.10.1
Warp CUDA error: Function cuCtxGetCurrent_f: a suitable driver entry point was not found
Warp CUDA error 36 (W:\a9ae500d09a78409\warp\native\warp.cu:155)
Traceback (most recent call last):
  File "E:\PycharmProjects\testSimulation\main.py", line 902, in <module>
    initScene()
  File "E:\PycharmProjects\testSimulation\main.py", line 567, in initScene
    cloth = Cloth(clothY, clothNumX, clothNumY, clothSpacing, sphereCenter, sphereRadius)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\PycharmProjects\testSimulation\main.py", line 115, in __init__
    self.pos = wp.array(pos, dtype=wp.vec3, device="cuda")
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\PycharmProjects\testSimulation\venv\Lib\site-packages\warp\types.py", line 1273, in __init__
    self._init_from_data(data, dtype, shape, device, copy, pinned)
  File "E:\PycharmProjects\testSimulation\venv\Lib\site-packages\warp\types.py", line 1396, in _init_from_data
    device = warp.get_device(device)
             ^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\PycharmProjects\testSimulation\venv\Lib\site-packages\warp\context.py", line 2500, in get_device
    return runtime.get_device(ident)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\PycharmProjects\testSimulation\venv\Lib\site-packages\warp\context.py", line 2323, in get_device
    return self.get_current_cuda_device()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\PycharmProjects\testSimulation\venv\Lib\site-packages\warp\context.py", line 2354, in get_current_cuda_device
    raise RuntimeError("CUDA is not available")
RuntimeError: CUDA is not available
```

c0d1f1ed, 2023-12-04:
Sorry about the unclear error message. The GeForce GT 730M is unfortunately not supported by Warp. We require [CUDA Compute Capability](https://developer.nvidia.com/cuda-gpus) 5.2 as a minimum, and the GT 730M only supports 3.0. Warp uses CUDA Tookit 11.5 or higher, which [no longer supports](https://en.wikipedia.org/wiki/CUDA#GPUs_supported) GPUs with Compute Capability 3.0.

We'll improve our documentation and the error message in a future release.

c0d1f1ed, 2024-03-05:
https://github.com/NVIDIA/warp/commit/d43b12c2ba2afd33580719adb7b10407f6195fd3 improves the error reporting and documentation.

robotNo201, 2024-12-27:
| NVIDIA-SMI 560.35.03              Driver Version: 561.09         CUDA Version: 12.6
nvcc version 11.8
torch version 2.4.0+cu118
this won't work either