NVIDIA/warp #170: Boids example not loading

cadop, 2023-12-21:
I just installed 2022.2.2. Clicked on the Examples, press play, nothing happens. 

[image]

I think the example is just the action graph and is missing the scene setup (e.g. `2023-12-21 00:07:51  [Warning] [omni.graph.core.plugin] Could not find node type interface for 'omni.graph.action_nodes.OnLoaded'`

cadop, 2023-12-21:
When adding a Tick in the graph, it asks for an updated warp 

[image]

after that, it asks to updated usdrt 
[image]

But there doesnt seem to be that option in 2023.2.2 beta

christophercrouzet, 2023-12-21:
Hi @cadop , this specific sample indeed requires a version of Composer USD that isn't released yet, so it won't run on current versions. We need to make this clearer somehow. Apologies for the inconvenience caused!

cadop, 2023-12-21:
I guess it shouldn't have been included in the release?  Can the release of warp get updated to remove the example? It's not possible to run it all, so kinda weird.

shi-eric, 2024-06-06:
@christophercrouzet Is this fixed now?

christophercrouzet, 2024-06-06:
It should work with Kit 106.0, which has just been released, but USD Composer hasn't been updated yet.

cadop, 2024-06-18:
I can open the usda file directly in IsaacSim 4.0 and it works. 

A note though; the link on the graph object is broken. 

[image]

Goes to: https://docs.omniverse.nvidia.com/extensions/latest/ext_omnigraph/node-library/nodes/omni-warp/warpsampleprimflocking-1.html

christophercrouzet, 2024-06-23:
Thanks @cadop! This seems to be the case for other nodes as well (not only Warp ones), I'll report this to the corresponding team.

closed 2024-07-09 by shi-eric