NVIDIA/warp #173: particle delta calculations are constraints/(w1+w2) but should be constraints*w1/(w1+w2)?

yingDaWang-UNSW, 2024-01-08:
Hi warp team, I noticed while testing xpbd in a simple system of particles with contrasting radius and masses, that unexpected results relating to delta calculations were occurring. 

I note that in the xpbd integrator, for particle-ground, particle-shape, and particle-particle kernels, the deltas are                 delta += (delta_f - delta_n) / denom    or of a similar nature. 

If I make the following change to the kernels:  delta += (delta_f - delta_n) / denom * w1.

Then both gravity segregation and the brazil nut effect emerge. Without this, the high w particles have deltas that are too low and vice versa.

eric-heiden, 2024-04-02:
Hi @yingDaWang-UNSW,

Thank you for reporting this issue! We have recently updated the XPBD integrator with these particle contact constraint modifications.

shi-eric, 2024-06-06:
I believe this was fixed by @eric-heiden in 302b79e37725ad46eed4b49b0ba6d4d5160e0dda. Please reopen this issue if this is not the case. Thanks!

closed 2024-06-06 by shi-eric