NVIDIA/warp #176: CUDA Compilation errors when using the new interface to wp.mesh_query_ray()

maria-korosteleva, 2024-01-17:
Hi!

I'm trying to use a freshly introduced interface to ` wp.mesh_query_ray()` inside a kernel, running on CUDA. 
The code snippet looks like this: 
```
out = wp.mesh_query_ray(mesh_id, origin, direction, max_t)
if out.result:
    <do something>
```

Once I run it, I get the following errors: 

```
default_program(26126): error: a reference of type "float &" (not const-qualified) cannot be initialized with a value of type "wp::mesh_query_ray_t"
          wp::adj_mesh_query_ray(var_6, var_29, var_34, var_33, var_35, adj_6, adj_29, adj_34, adj_33, adj_35);
                                                                ^

default_program(26126): error: a reference of type "float &" (not const-qualified) cannot be initialized with a value of type "wp::uint64"
          wp::adj_mesh_query_ray(var_6, var_29, var_34, var_33, var_35, adj_6, adj_29, adj_34, adj_33, adj_35);
                                                                        ^

default_program(26126): error: a reference of type "float &" (not const-qualified) cannot be initialized with a value of type "wp::vec_t<3U, float>"
          wp::adj_mesh_query_ray(var_6, var_29, var_34, var_33, var_35, adj_6, adj_29, adj_34, adj_33, adj_35);
                                                                               ^

default_program(26126): error: a reference of type "float &" (not const-qualified) cannot be initialized with a value of type "wp::vec_t<3U, float>"
          wp::adj_mesh_query_ray(var_6, var_29, var_34, var_33, var_35, adj_6, adj_29, adj_34, adj_33, adj_35);
                                                                                       ^

default_program(26126): error: a reference of type "wp::vec3 &" (not const-qualified) cannot be initialized with a value of type "wp::float32"
          wp::adj_mesh_query_ray(var_6, var_29, var_34, var_33, var_35, adj_6, adj_29, adj_34, adj_33, adj_35);
                                                                                               ^

default_program(26126): error: a reference of type "int &" (not const-qualified) cannot be initialized with a value of type "wp::mesh_query_ray_t"
          wp::adj_mesh_query_ray(var_6, var_29, var_34, var_33, var_35, adj_6, adj_29, adj_34, adj_33, adj_35);
                                                                                                       ^

default_program(26126): error: too few arguments in function call
          wp::adj_mesh_query_ray(var_6, var_29, var_34, var_33, var_35, adj_6, adj_29, adj_34, adj_33, adj_35);

```

Which then finishes with the exception:   
```
File "C:\Users\mariako\Documents\Code\garmentsimwarp\warp\sim\collide.py", line 2009, in collide
    wp.launch(
  File "C:\Users\mariako\Documents\Code\garmentsimwarp\warp\context.py", line 3535, in launch
    if not module.load(device):
  File "C:\Users\mariako\Documents\Code\garmentsimwarp\warp\context.py", line 1654, in load
    raise (e)
  File "C:\Users\mariako\Documents\Code\garmentsimwarp\warp\context.py", line 1632, in load
    warp.build.build_cuda(
  File "C:\Users\mariako\Documents\Code\garmentsimwarp\warp\build.py", line 30, in build_cuda
    raise Exception(f"CUDA kernel build failed with error code {err}")
Exception: CUDA kernel build failed with error code 6`

```

If I use the old interface, the compilation and computation works fine: 

```
if wp.mesh_query_ray(body_mesh, o, d, max_t, t, face_u, face_v, sign, normal, face_index):
    <do something>

```
Looks like a bug was introduced with the new interfaces 

**Setup:**
(on Windows 11)
Warp 1.0.0-beta.6 initialized:
   CUDA Toolkit: 12.3, Driver: 12.3
   Devices:
     "cpu"    | Intel64 Family 6 Model 151 Stepping 2, GenuineIntel
     "cuda:0" | NVIDIA GeForce RTX 3090 Ti (sm_86)

christophercrouzet, 2024-01-17:
Hi @maria-korosteleva, thanks for reporting this issue!

This is actually due to a silly oversight on my end and has since been addressed, so the next release of Warp will work as expected.

In case you don't need differentiability for this specific kernel, you could also decorate it with `@wp.kernel(enable_backward=False)`, which avoids this issue (and might save a bit in compilation time).

Apologies for any inconvenience caused!

maria-korosteleva, 2024-01-22:
@christophercrouzet Got it. Thank you for quick response!

c0d1f1ed, 2024-03-05:
This was addressed by https://github.com/NVIDIA/warp/commit/9316c2176d447440fd626c6add8e1e010bd846ad. Thanks Chris!

closed 2024-03-05 by c0d1f1ed